AdamWelchUK/NodeRedEPEverDashboard #6: Many decimal

WingChunBR, 2021-05-17:
Sorry for posting as an issue (must be my config) and for my poor english writing
Thank you for the dashboard. I've been using it for an year. Since first config, I noticed it put more decimal (places?) at random, every refresh of data from solar controllers.
I assume it's should be rounded to 2 decimal by looking at config. May be a raspberry nodered problem?

[image]
[image]

dtabone, 2021-05-17:
Try setting the Value Format as in the picture below:
[image]

WingChunBR, 2021-05-18:
thank you @dtabone 
fixed in my case
Is it a common problem? don't know how to submit a "request" for change in the code

AdamWelchUK, 2021-05-18:
I think this is caused by the range node, but as shown it can be rectified by the chart node - thanks dtabone.
I have been playing around with an alternative version of the flows which uses more function nodes and the javascript which seems to be better at dividing by 100! Let me know if you want to see the alternatives.