AdamWelchUK/NodeRedEPEverDashboard #7: Include charging status

WingChunBR, 2021-05-18:
Tried for hours to get information from this modbus register data. Couldn't find on other projects too.
Would be useful to include at least D3-2 (show the current status of charging). SOC shows 100% even at beginning of charge process (voltage calculated, I guess). I'm trying to think a more precise calculation using other methods

[image]

AdamWelchUK, 2021-05-18:
Hello. Yeah I remember having a little go with that register, but not getting very far. I've just had another look and I can't work out what the response I am getting means.
Not sure what to say...

WingChunBR, 2021-05-22:
I'm glad to know, as a totally node-red noob, that the register response is not so easy to work

dtabone, 2021-06-16:
Updated JSON to include battery stats - see attached
[image]

[EPEverDashboardv0.3.zip](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/6660978/EPEverDashboardv0.3.zip)