AdamWelchUK/NodeRedEPEverDashboard #8: Updated v0.4 Dashboard

dtabone, 2021-06-17:
- Fixed some field types for Battery Status
- Added device data & time (read only for now)
-
[EPEverDashboardv0.4.json.zip](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/6668038/EPEverDashboardv0.4.json.zip)

AdamWelchUK, 2021-06-17:
Ace - seems to work for me. Thank you.
I've been playing with a version that uses more javascript in function blocks, its tidier to look at, but may be harder for some to understand how it works. Not sure what you think about that?
[image]

dtabone, 2021-06-17:
Looking good! Function blocks are def the way to go. Im and old school
software engineer but i understand the concepts and used functions to
employ bitwise operations. Feel free to reuse what i shared in your own
version .. bottom line is usefulness for me.

dtabone, 2021-06-25:
Updated JSON v0.5 attached - tested on TriRon3210 V01.56+V01.22j

- separated the time + date from main settings read/write
- added separate read/write date+time from/to controller

[EPEverDashboardv0.5.json.zip](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/6717666/EPEverDashboardv0.5.json.zip)

Laurieralma, 2021-07-16:
hi adam, i have a (eperver xtra 3215 with ebox01 wifi) i installed on raspberry pi 4, your eperver dashboard .5 json.  I have a camp in the forest without internet.  could you tell me how to configure (modbus get) node to see my xtra 3215. thank you