AdamWelchUK/NodeRedEPEverDashboard #9: Add current time...

SCarns, 2021-06-25:
Looking over this and it seems to make sense to update the unit's time and date when writing out the settings data.

SCarns, 2021-06-25:
I'm looking to add a 3rd flow in "Settings" to run once a day to send current time & date to the controller. To make this work, I need to figure out how to get date/time into a format Modbus can receive. The rest is easy, as I can just copy the row "2" above to write values for [9013, 9014, 9015];

SCarns, 2021-06-25:
I'm stuck here:

SCarns, 2021-06-25:
[{"id":"6e05060.f1027fc","type":"tab","label":"Commander Settings","disabled":false,"info":""},{"id":"f9ae754c.577d5","type":"modbus-flex-getter","z":"6e05060.f1027fc","name":"Modbus Get","showStatusActivities":false,"showErrors":false,"logIOActivities":false,"server":"c082a42f.1ec8e","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"keepMsgProperties":false,"x":210,"y":120,"wires":[["7895ffde.47f8f8"],["304fc50f.822082"]]},{"id":"519b5a30.ab867c","type":"function","z":"6e05060.f1027fc","name":"FC3 0x9000 [15]","func":"msg.payload = { \n    'fc': 3, \n    'unitid': 1, \n    'address': 0x9000, \n    'quantity': 15\n}\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":140,"y":180,"wires":[["f9ae754c.577d5"]]},{"id":"304fc50f.822082","type":"modbus-response","z":"6e05060.f1027fc","name":"0x9000 Response","registerShowMax":20,"x":350,"y":20,"wires":[]},{"id":"2a2ec45f.49aa04","type":"switch","z":"6e05060.f1027fc","name":"","property":"parts.index","propertyType":"msg","rules":[{"t":"eq","v":"0","vt":"num"},{"t":"eq","v":"1","vt":"num"},{"t":"eq","v":"2","vt":"num"},{"t":"eq","v":"3","vt":"str"},{"t":"eq","v":"4","vt":"str"},{"t":"eq","v":"5","vt":"str"},{"t":"eq","v":"6","vt":"str"},{"t":"eq","v":"7","vt":"str"},{"t":"eq","v":"8","vt":"str"},{"t":"eq","v":"9","vt":"str"},{"t":"eq","v":"10","vt":"str"},{"t":"eq","v":"11","vt":"str"},{"t":"eq","v":"12","vt":"str"},{"t":"eq","v":"13","vt":"str"},{"t":"eq","v":"14","vt":"str"}],"checkall":"false","repair":false,"outputs":15,"x":350,"y":280,"wires":[["3eb2f325.078bb4"],["8080685f.2ebbb8"],["f3c2d559.c1ba4"],["c0a9a722.12e4a"],["b01fc8f9.79b7c8"],["ae1677c1.2051c"],["5fb74957.5d3738"],["3b619bd9.e0fe4c"],["759737ef.49f3b8"],["9b1be14f.ce0d98"],["75c32c8d.868d74"],["ca8fb86e.e36b2"],["f8b26a58.bec238"],["7cc59176.8db3d"],["5ea833e8.7b3cac"]]},{"id":"c0a9a722.12e4a","type":"range","z":"6e05060.f1027fc","minin":"0","maxin":"10000","minout":"0","maxout":"100","action":"scale","round":false,"property":"payload","name":"/100","x":530,"y":260,"wires":[["2a03600b.24ca48"]]},{"id":"7895ffde.47f8f8","type":"split","z":"6e05060.f1027fc","name":"","splt":"\\n","spltType":"str","arraySplt":1,"arraySpltType":"len","stream":false,"addname":"","x":230,"y":280,"wires":[["2a2ec45f.49aa04"]]},{"id":"2a03600b.24ca48","type":"ui_numeric","z":"6e05060.f1027fc","name":"","label":"High Voltage Disconnect","tooltip":"","group":"6940fb1e.3ea9f4","order":4,"width":0,"height":0,"wrap":false,"passthru":true,"topic":"","format":"{{value}} V","min":"9","max":"17","step":"0.1","x":710,"y":260,"wires":[["5a7c9ceb.b0cff4"]]},{"id":"b01fc8f9.79b7c8","type":"range","z":"6e05060.f1027fc","minin":"0","maxin":"10000","minout":"0","maxout":"100","action":"scale","round":false,"property":"payload","name":"/100","x":530,"y":320,"wires":[["1e1ca9ef.bc6136"]]},{"id":"1e1ca9ef.bc6136","type":"ui_numeric","z":"6e05060.f1027fc","name":"","label":"Charging Limit","tooltip":"","group":"6940fb1e.3ea9f4","order":5,"width":0,"height":0,"wrap":false,"passthru":true,"topic":"","format":"{{value}} V","min":"9","max":"17","step":"0.1","x":680,"y":320,"wires":[["1ed3292d.359f4f"]]},{"id":"ae1677c1.2051c","type":"range","z":"6e05060.f1027fc","minin":"0","maxin":"10000","minout":"0","maxout":"100","action":"scale","round":false,"property":"payload","name":"/100","x":530,"y":380,"wires":[["5638ccf5.2f9434"]]},{"id":"5638ccf5.2f9434","type":"ui_numeric","z":"6e05060.f1027fc","name":"","label":"Over Voltage Reconnect ","tooltip":"","group":"6940fb1e.3ea9f4","order":6,"width":0,"height":0,"wrap":false,"passthru":true,"topic":"","format":"{{value}} V","min":"9","max":"17","step":"0.1","x":710,"y":380,"wires":[["a49b57c2.da7248"]]},{"id":"86f6460e.e872e","type":"ui_button","z":"6e05060.f1027fc","name":"","group":"6940fb1e.3ea9f4","order":1,"width":0,"height":0,"passthru":true,"label":"Read Values","tooltip":"Click to read currently set values.","color":"","bgcolor":"","icon":"","payload":"1","payloadType":"num","topic":"","x":150,"y":540,"wires":[["519b5a30.ab867c","8c2f0620.78ea08","fdd4e8c9.9ad558"]]},{"id":"bb027f4b.90d2d","type":"ui_numeric","z":"6e05060.f1027fc","name":"","label":"Equalization","tooltip":"","group":"6940fb1e.3ea9f4","order":7,"width":0,"height":0,"wrap":false,"passthru":true,"topic":"","format":"{{value}} V","min":"9","max":"17","step":"0.1","x":670,"y":440,"wires":[["e2272154.a04dc"]]},{"id":"3b619bd9.e0fe4c","type":"range","z":"6e05060.f1027fc","minin":"0","maxin":"10000","minout":"0","maxout":"100","action":"scale","round":false,"property":"payload","name":"/100","x":530,"y":500,"wires":[["a4cc09bb.84289"]]},{"id":"a4cc09bb.84289","type":"ui_numeric","z":"6e05060.f1027fc","name":"","label":"Boost","tooltip":"","group":"6940fb1e.3ea9f4","order":8,"width":0,"height":0,"wrap":false,"passthru":true,"topic":"","format":"{{value}} V","min":"9","max":"17","step":"0.1","x":650,"y":500,"wires":[["6378392a.1e46d8"]]},{"id":"759737ef.49f3b8","type":"range","z":"6e05060.f1027fc","minin":"0","maxin":"10000","minout":"0","maxout":"100","action":"scale","round":false,"property":"payload","name":"/100","x":530,"y":560,"wires":[["18f4f8cb.894617"]]},{"id":"18f4f8cb.894617","type":"ui_numeric","z":"6e05060.f1027fc","name":"","label":"Float","tooltip":"","group":"6940fb1e.3ea9f4","order":9,"width":0,"height":0,"wrap":false,"passthru":true,"topic":"","format":"{{value}} V","min":"9","max":"17","step":"0.1","x":650,"y":560,"wires":[["7beed0a8.b4a768"]]},{"id":"9b1be14f.c [... cut ...] 600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"","commandDelay":"100","clientTimeout":"2000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false},{"id":"64dde05.682662","type":"ui_group","name":"Solar","tab":"6d3a1d1a.0ff764","order":1,"disp":true,"width":"6","collapse":false},{"id":"6d3a1d1a.0ff764","type":"ui_tab","name":"EPEver Dashboard","icon":"dashboard","order":2,"disabled":false,"hidden":false}]

dtabone, 2021-06-25:
You need to use reassemble the vars and use bitwise conversions to create the array. I'm almost done doing it,

dtabone, 2021-06-25:
@SCarns  see the previous issue - i've uploaded the JSON v0.5 zip file