AdamWelchUK/NodeRedEPEverDashboard #10: EPSolar Dashboard Error

kasper58, 2021-09-16:
Hi Adam, 
Ran into a problem with installing your Dashboard into Node-Red. 
Import into clipboard gives an "ERROR: Input not a JSON Array."
See attached screenshot.
[image]

Regards Warren

kasper58, 2021-09-16:
Duh..... Sorted thanks
The file... not the filename.

kasper58, 2021-09-16:
- [ ] Duh..... Sorted thanks
The file... not the filename.

closed 2021-09-16 by kasper58

dtabone, 2021-09-16:
Kasper58 See this post too.. https://github.com/AdamWelchUK/NodeRedEPEverDashboard/issues/8#issuecomment-868697479

kasper58, 2021-09-17:
> Kasper58 See this post too.. #8 (comment)

Thanks Adam.
I saw that and are trying to use it as a basis for my EPEver Charger /
Inverter. Having some issues getting RS458 to connect.
I have the protocol  sheet (att) but nobody is at home.

Regards Warren