AdamWelchUK/NodeRedEPEverDashboard #12: Inconsistent values on solar & batt power?

emiura, 2021-11-08:
I have started using you dashboard today, but I wonder if the graphs for solar & batt power need some adjustment, because sometimes they appear to report a value way lower than the values on left panel.
For example, sometimes my generation hits values higher than 500W, but the graph show values like 30 or 50W.
[image]

jphein, 2021-11-15:
Same thing happened to me, everytime power was generated above a certain level. I ended up just multiplying volts by amps instead. I think it is a Tracer - AN issue? Maybe a power factor thing? Just guessing.

AdamWelchUK, 2021-11-17:
Solar power is calculated using two 16bit modbus registers and my solar panels don't generate enough power to overflow and use the higher register. This was one of the bits I could never really check I'm afraid
If you can capture a modbus response for me (like my 'fake stats for testing' string) and let me know the value you think should be shown, I can try edit the code to suit.
Eg. [ 1357, 0, 50, 0, 1275, 5, 63, 0, 1275, 5, 63, 0, 756, 1470 ]  13.5 watts
Or else you could start playing around with the code in the '32bit register conversion /100' node.