AdamWelchUK/NodeRedEPEverDashboard #13: Not working with XTRA series in combination with the ebox-Wif-01

sdavidst, 2021-12-26:
I boughtan Epever XTRA3210N charge controller and connect it to my node-red environment with an ebox-Wif-01. 
Unfortunately I can't get it to work!!!!  I put the IP-address of the ebox-wifi as a server in the modbus -Flex-getter node. It just don't get any answers back.
Maybe it doesn't work because the XTRA series are different?
Maybe I need a virtual com port (in my case a linux port).
I have tried everything.
Is there someone who can help me, did somebody with an XTRA series controller get it to work???

Laurieralma, 2021-12-27:
[image]
i have an epever xtra 3215 and the evever wifi-01 works fine with it.

sdavidst, 2021-12-27:
Thank you so much Laurieralma, I changed to the telnet option and now I see that I get answers from the controller. You made my day!

closed 2021-12-27 by sdavidst