AdamWelchUK/NodeRedEPEverDashboard #14: WiFi Dashboard with eBox-WIFI-01 (directions for relaying Serial connection)

whitneykrape, 2021-12-31:
Hiya, 

First, massive thank you. This library/setup is vastly better than the "official" apps for the eBox module. The directions below were pulled from a few places and figured newer folks might get the eBox-WIFI-01 module expecting plug and play like I did only to find a massive maze. Might be good for an addendum readme.

Note: If you haven't bought the "official" eBox WiFi module yet, consider the alternatives first, other options are more DIY but offer better setup and more options. Skip to #4 for the Serial connection directions.
1. Attach module and directly connect via browser to eBox-WIFI-01.
2. If the "STA Settings" tab is not available the loaded firmware might be more restrictive. Search "eBox-WIFI-01 sta mode" and troubleshoot accordingly if you want the WiFi to connect to a local network. Once you have the IP for the eBox, note for later. 
3. Install Node-Red and add the EPEverDashboard as described to desired monitor system (Pi in this example.) 
4. The eBox WiFi module is actually a simple Wireless to Serial connection but lacks translation in it's output. On the Pi (or any other device) we need to bridge the Serial connection to the Wireless on the monitoring device. To do this use `socat` as a relay:
	`sudo socat pty,link=/dev/ttyWiFi211,unlink-close=0,raw,echo=0 tcp:[ipofeboxwifi]:8088&`
	(Adjust IP, Port, Directory as needed.) 
	[Source](http://randomsporadicprojects.blogspot.com/2016/11/instructions-for-creating-dashboard-to_9.html)
5. Set the permissions of the link directory. 
	`sudo chmod 777 /dev/ttyWiFi211` 
	(These are very open permissions, adjust as security as needed.)
6. In the Node-Red EPEverDashboard select the modbus-client -> TracerA12 setting and change the Serial Port setting to the link directory created before, `/dev/ttyWiFi211`
7. Deploy and check the UI, everything should populate and be able to Read/Write the battery values.

tbodor, 2022-01-12:
This seems to work, too, only very limited testing so far, though. 
[image]

whitneykrape, 2022-01-17:
Oh good call, I didn't see that before. Just tried it and did get it working with the TCP TELNET, the other options (don't totally know what they are for) don't seem to connect.