AdamWelchUK/NodeRedEPEverDashboard #17: New dashboard with several modifications such as rounding values ​​to 2 digits after the dot.  see daily, monthly, annual, total kwh generated and consumed.  etc.  and in French

Laurieralma, 2022-03-14:
New dashboard with several modifications such as rounding values ​​to 2 digits after the dot.  see daily, monthly, annual, total kwh generated and consumed.  etc.  and in French .csv file to improve see examples Excel file.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[dashboard .6 fr.json.txt](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/8244993/dashboard.6.fr.json.txt)

dtabone, 2022-03-14:
Thank you for sharing :)

VincentPestana, 2022-04-16:
@Laurieralma Can I please start a new repo with your dashboard in it? I checked your Github and you dont have a Repo for it.

Also here is a small fix, your "Solaire Voltage" graph was set to 7 days, where as the rest are set to 2 days
[dashboard.6.fr.json.txt](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/8499445/dashboard.6.fr.json.txt)

Laurieralma, 2022-04-17:
all voltage, current or wattage nodes are adjustable according to your needs, I took 7 days for the voltage of the solar panels to test with 7 days.
[image]

Laurieralma, 2022-04-17:
New csv construction 
[image]

Laurieralma, 2022-04-17:
[flows csv construction avril2022.json.txt](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/8500808/flows.csv.construction.avril2022.json.txt)

Laurieralma, 2022-04-17:
He creates a daily.csv and monthly backup with the date and resets the stats.csv file at the beginning of the month because it becomes too large and practically unusable.  It takes a recording every 15 minutes to make it more flexible with exel and making charts it take quandary even more than 3000 readings per month!