AdamWelchUK/NodeRedEPEverDashboard #18: Problem with configuration modbus-client

geophys-hub, 2022-03-20:
Hello Adam,
thank you for your hard work.
I have a connection problem.
I am using epever tracer 3210an, a RS485 USB cable and a RASPI 3b. I installed Node-Red yesterday following your instructions. But unfortunately I can't get in touch with EPEver.
Here is the event in /var/log/daemon.log: 
{
Mar 20 16:00:50 rpi3 Node-RED[410]:   payload: {
Mar 20 16:00:50 rpi3 Node-RED[410]:     value: undefined,
Mar 20 16:00:50 rpi3 Node-RED[410]:     unitid: 1,
Mar 20 16:00:50 rpi3 Node-RED[410]:     fc: 4,
Mar 20 16:00:50 rpi3 Node-RED[410]:     address: 12544,
Mar 20 16:00:50 rpi3 Node-RED[410]:     quantity: 18,
Mar 20 16:00:50 rpi3 Node-RED[410]:     emptyMsgOnFail: true,
Mar 20 16:00:50 rpi3 Node-RED[410]:     keepMsgProperties: true,
Mar 20 16:00:50 rpi3 Node-RED[410]:     messageId: new ObjectId("623741a2fcca8f991afd26ab")
Mar 20 16:00:50 rpi3 Node-RED[410]:   },
Mar 20 16:00:50 rpi3 Node-RED[410]:   topic: 'livestats',
Mar 20 16:00:50 rpi3 Node-RED[410]:   _msgid: 'e897b8b8ad8aa219',
Mar 20 16:00:50 rpi3 Node-RED[410]:   messageId: new ObjectId("623741a2fcca8f991afd26ab"),
Mar 20 16:00:50 rpi3 Node-RED[410]:   error: {
Mar 20 16:00:50 rpi3 Node-RED[410]:     message: 'Error: Client Not Ready To Read At State reconnecting',
Mar 20 16:00:50 rpi3 Node-RED[410]:     source: {
Mar 20 16:00:50 rpi3 Node-RED[410]:       id: '2c18fdaf.8153d2',
Mar 20 16:00:50 rpi3 Node-RED[410]:       type: 'modbus-flex-getter',
Mar 20 16:00:50 rpi3 Node-RED[410]:       name: 'Modbus Get',
Mar 20 16:00:50 rpi3 Node-RED[410]:       count: 1
Mar 20 16:00:50 rpi3 Node-RED[410]:     },
Mar 20 16:00:50 rpi3 Node-RED[410]:     stack: 'Error: Client Not Ready To Read At State reconnecting\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object.<anonymous> (/home/pi/.node-red/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:472:15)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object.emit (events.js:314:20)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object.emit (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:185:20)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object._inputCallback (/home/pi/.node-red/node_modules/node-red-contrib-modbus/modbus/maps/modbus-flex-getter.js:146:21)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:207:26\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object.trigger (/usr/lib/node_modules/node-red/node_modules/@node-red/util/lib/hooks.js:149:13)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object.Node._emitInput (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:199:11)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object.Node.emit (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:183:25)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Object.Node.receive (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:482:10)\n' +
Mar 20 16:00:50 rpi3 Node-RED[410]:       '    at Immediate._onImmediate (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/flows/Flow.js:814:52)'
Mar 20 16:00:50 rpi3 Node-RED[410]:   }
Mar 20 16:00:50 rpi3 Node-RED[410]: }

***********************

I configured the modbus clients as follows: 

[image]

My cable is recognized, a driver is loaded.
I checked that with lsusb and in dmesg.

Can you spot an error in the screenshot?

Many thanks in advance 

Frank

kedar-1, 2022-03-20:
I have the same tracer.
You must use  xr_usb driver.
Follow the step by step guide: https://stackoverflow.com/questions/57313343/modbus-cant-read-data-from-epever-tracer-1210a-to-raspberry

geophys-hub, 2022-03-21:
That also makes me think...

[image]

geophys-hub, 2022-03-21:
[image] Do I have the correct cable to read out?

geophys-hub, 2022-03-21:
PROBLEM is SOLVED!

The problem sat 30 cm in front of the screen! I had defined the device /dev/ttyRXUSB0. That was incorrect! /dev/ttyXRUSB0 is correct ;)

[image]

Now, the testscript works fine

Laurieralma, 2022-03-21:
[image]
Wich std rj45 pin 4 blue on -, pin 6 green on +
[image]

geophys-hub, 2022-03-21:
Thanks! 
But i have the cable from the manual ... CC...150

geophys-hub, 2022-03-21:
If I don't find it I'll reinstall the raspi image and node-red too.

The real problem was actually the missing XR driver as Adam wrote.

kedar-1, 2022-03-22:
Change the type of Modbus server to "Serial" from "Serial Expert". 
My setup is working without modbus read issues over six months.

mmsfla, 2022-03-28:
The problem could be changes in the register addresses and function code of the newly purchased EPEver Tracer 3210AN.

I found references to this while researching.

If someone has the same problem, they can have a look here: 

https://github.com/rosswarren/epevermodbus

After installing you find in folder ./epevermodbus driver.py. 
Compares the parameters used there.

mitchross, 2022-07-23:
@mmsfla  if you are still stuck I wrote up a step by step foolproof guide. https://medium.com/@mitchross09/solar-energy-monitoring-with-raspberry-pi-and-node-red-ed59e287cdd4

TalShaf93, 2023-06-05:
for some reason after I remove the CDC-ACM driver and installing the driver I can't find the /dev/ttyXRUSB0. I tried it many times in different orders but same result.

anyone can help?

thanks in advance

[terminal image with no /dev/ttyXRUSB0](https://i.stack.imgur.com/jNVaO.png)