AdamWelchUK/NodeRedEPEverDashboard #19: Flow not working

kishfash, 2022-03-22:
Hi @AdamWelchUK ,

Thanks for the great work. 

Please I need help to get the flow to work. I'm using the original Epever USB to RS485 cable. I have installed the cable driver and set the Modbus Get but the flows seems not working.  I have previous used your EPEverDashboardv0.1.json with my old Raspberry Pi with same cable but can't find the old .json code again to try. Please I need help. 

Thank you.

Femi

PS:
I'm not a programmer but can follow instruction to get things work.

Attached is the screenshot of my setting.
[image]

[image]

kedar-1, 2022-03-23:
Try to run a Python code to verify the driver/cable is working as expected.

kishfash, 2022-03-23:
Hello @kedar-1,

Thank you for your reply. Kindly help with sample Python code to test.

Thank you.

closed 2022-03-25 by kingfash