AdamWelchUK/NodeRedEPEverDashboard #21: EPEver IP-3000 Inverter

Wizzkidy, 2022-03-31:
Hi,

I am using this great dashboard for two of my AN controllers and it works really well but I want to be able to get some data from my IP-3000 inverter.

Ideally I would like to pull the Output Power(W) into nodered and then if it drops to 0 (if my system switches back to mains) setup some kind of email notification.

I just can't find anything about monitoring this Epever IP-3000 inverter anywhere. Any help?

Wizzkidy, 2022-04-05:
Hi, Thanks for the reply. Yes I have contacted EPEver and managed to get the registers for this inverter model but I need some help putting this into nodered as I am a real novice with this stuff. I have attached the document they sent through
[Modbus Protocols for Inverter V1.0-EN.pdf](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/8419511/Modbus.Protocols.for.Inverter.V1.0-EN.pdf)

kedar-1, 2022-04-20:
You can replace/adapt registers from the Dashboard to satisfy requirements for inverter.

Wizzkidy, 2022-06-10:
So I got this working for most stats, but for Output Watts on the inverter, when it goes over 650watts the numbers all mess up and it then reports a wrong wattage.

I am unsure why this is happening.

TerryFrench, 2022-06-10:
Usually it’s due to the data being transmitted and being over one register. 
I.e. if it overflows one register it might be using 2. Then the code would need to query both from the device and do a bit of math to build a number based on both registers. 

I don’t have time to look but your inverter pdf should have the info. 
Best is likely to log the data you get. 
Do it with a light bulb and repeat with something using lots of power.

Wizzkidy, 2022-06-11:
Hi, Thanks although it doesn't really help me. The PDF does have a L and a H value for output watts, but I have no clue what code or what maths I need to perform to get the correct value. 

Might have to give up on this I just don't have the experience to get it working, I was able to get normal ones working that only require one code

AdamWelchUK, 2022-06-24:
Hi Wizzkidy.
Sorry I've not been very active, but thanks for trying out the dashboard. Also thanks to those who have been trying to help.

I've thrown this attached flow together which may help you.

This test flow should request the two registers which report load power from your inverter and how I think they should be processed to give you the right value. Let me know how you get on as I haven't been able to test it!

The other way to fix this would be to use the two stats you already got successfuly - voltage and current and multiply them to get power

[flows.zip](https://github.com/AdamWelchUK/NodeRedEPEverDashboard/files/8975976/flows.zip)

Wizzkidy, 2022-06-24:
Excellent this works perfectly! Thanks for this Adam, I have it all going into influx now.

Thanks again for your amazing work. Glad your back! 👍

closed 2022-07-03 by Wizzkidy