AdamWelchUK/NodeRedEPEverDashboard #22: Two Epever on one RS485?

Emmpunkt, 2022-07-21:
Hi,
is it possible to use one USB RS485 Adapter with two Epever in parallel?
I trying it for several hours now, but i´m failing ;)
My Raspi is not even starting if i don´t unplug the USB..
Emmpunkt

TerryFrench, 2022-07-22:
You might be able to change the device address of the 2 Epever. By default
they might have the same address.
With the LCD remote display in the settings, you can do that.
I'm not saying this will work, just that this might be a prerequisite.

Emmpunkt, 2022-07-22:
They both have different addresses. 1&2. I made this with the Epever
Software.

TerryFrench, 2022-07-31:
turn on logging / trace on the node getting the data.
Also depending on your hardware and how you transfer the bits, you might be
able to get some logging there too (if you have a serial/485 to IP
conversion for instance).
If you see packets from both devices, it might be possible to change the
NodeRed code to split the replies from each device.
That would be my *guess* here.